rust-embedded-community/epd-waveshare #26: Support Color Displays?

jamesmunns, 2019-03-31:
I recently ordered a collection of the WaveShare displays, including:

* 4.2" Yellow/Black/White
* 4.2" Black/White
* 4.2" Red/Black/White
* 7.5" Black/White
* 5.83" Red/Black White

I'm happy to contribute or help test, as I plan to use these on a couple of embedded devices for my [internet of streams](https://github.com/ferrous-systems/internet-of-streams) project. I was wondering if you had any notes already on where I should be looking to help add support for the color displays.

caemor, 2019-04-03:
Sorry for the late response. 

- the 4.2" B/W is working already. 
- the 7.5" B/W should also be pretty easy to add (I also got one at home but didn#t have the time to add it and test it)

AFAIK the colored ones just use two buffers instead of one so that also shouldn't be too complicated too add (where the biggest problem will be how to add/change [color.rs](https://github.com/caemor/epd-waveshare/blob/master/src/color.rs) and if 2 small buffers are better or one big buffer). I am really unsure about this problem.

Question to the naming convention: the call it 4in2(b) but wouldn't 4in2_red or something similar be better?

### Steps that need to be done to add a display generally: (e.g for the 4in2 (b)/red one)

1. Get the **Command table** from the datasheet (e.g here: https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_(B)#Datasheets ) and add it to a command enum like here (https://github.com/caemor/epd-waveshare/blob/master/src/epd4in2/command.rs) in a epd4in2b folder. This can also be found in waveshares new github repo (e.g.: https://github.com/waveshare/e-Paper/blob/master/4.2inch_e-paper_b%26c_code/wiringpi/obj/EPD_4in2b.h but be aware of quite a lot of copy&paste errors like the wrong ifndef epd2in7b at the top and better crosscheck it once more)

2. Optional: Add the **LUT constants** (like https://github.com/caemor/epd-waveshare/blob/master/src/epd4in2/constants.rs) if you can find any. But since the color ones only allow the default refresh they are most likely not included in the repos. This is also not too important/optional.

3. Add following constants from the next four lines to epd4in2b/mod.rs (https://github.com/caemor/epd-waveshare/blob/master/src/epd4in2/mod.rs#L61): width, height, default_background_color (twice for red/black displays?), is_busy_low

4. Implement **WaveshareDisplay** and **InternalWiAdditions** trait for your EPD4in2_Red struct.
See https://github.com/caemor/epd-waveshare/blob/master/src/epd4in2/mod.rs#L76 and following for examples and crosscheck it with the waveshare repos for the differences between different devices

   The Traits that need to be implemented are explained here: https://github.com/caemor/epd-waveshare/blob/master/src/traits.rs

   Maybe some adoptions for either a twice as big sized buffer or double buffers need to be made.

   Communication is handled by the DisplayInterface and is working the same through all waveshare displays that I know besides that the busy_low pin is sometimes inverted.

   Now you can already draw to the display, but you might want to use the embedded_graphics ;-)

5. Implement the graphics traits (Drawing<Color> & Display) e.g. like here: https://github.com/caemor/epd-waveshare/blob/master/src/epd4in2/graphics.rs
    
    The graphics.rs file is handling most of the graphics stuff for you: 
https://github.com/caemor/epd-waveshare/blob/master/src/graphics.rs

    But most likely there also need to be done some changes for the multicolor display.

Further stuff:

- Most likely Red and Yellow Versions of the same size work completely the same and just show a different color (I still need to check that)

- Naming of function/traits: If you have better names for some of the traits/functions or think that some don't really match with what they do I am always happy with improvements. 

### Good references are: 
 - https://github.com/waveshare/e-Paper
 - wikipages of the different displays e.g. [4.2inch_Module_B](https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_(B)) 
 - https://github.com/ZinggJM/GxEPD
 - https://github.com/olikraus/u8g2

I am happy to help if you have more specific question but be aware that the colored displays are quite  slow.

marcelbuesing, 2019-04-26:
I just ordered a 4.2 B/W and a 7.5 Red/Black/White.

Has anyone started adding colored display support yet ? Otherwise I might also start looking into it once they arrive. @caemor thanks for those infos.

caemor, 2019-04-26:
I haven't started working on it yet. I might have some time on Sunday at the Oxidize Conf.

caemor, 2019-04-30:
I couldn't find any time on Sunday and gonna be travelling for the next 3 weeks, so you can just start if you want.

DanSGiesbrecht, 2019-12-16:
I ordered a R/B/W display as well - I may have a go at this *someday*.

caemor, 2019-12-28:
@jkristell  added support for the 1in54 r/b/w display here: https://github.com/caemor/epd-waveshare/pull/35 and has an example on using it here: https://github.com/jkristell/nucleo-f401re/tree/add-eink-example/examples

caemor, 2020-05-24:
Although multiple ones were added I am leaving this issue open since it also includes some documentation about how to add a display

Alexx-G, 2021-03-03:
Thanks a lot @caemor for your effort!
Is 3.7in b/w display supported? If no, is #35 still an accurate example of adding a new display?

jamesmunns, 2021-03-04:
I recently ordered a couple newer displays, including the 3.7in b/w display. I haven't tried bringing it up yet, but if you need a second person to test, feel free to ping me!

Alexx-G, 2021-03-04:
Thanks @jamesmunns !
I've started working on a PR yesterday, however I'm struggling finding a command table for 3.7 - https://www.waveshare.com/wiki/3.7inch_e-Paper_HAT

caemor, 2021-03-04:
> Thanks a lot @caemor for your effort!
> Is 3.7in b/w display supported?

Not yet. 
> If no, is #35 still an accurate example of adding a new display?

#35 is is a red/black/white display so it is slightly different. 
I believe the instructions from above might be more helpful but #35 as well as any of the other recent PRs adding a new display can help you, just be aware that your display differs in the colors. 

Since your 3in7 b/w display seems to have some gray tones you need to define a new color type in color.rs similar to what is defined in their wiki: https://www.waveshare.com/wiki/3.7inch_e-Paper_HAT 

So while finishing this I saw you already added a new comment :-D

> Thanks @jamesmunns !
I've started working on a PR yesterday, however I'm struggling finding a command table for 3.7 - https://www.waveshare.com/wiki/3.7inch_e-Paper_HAT

If you click on ressources you can find the datasheet. (e.g. Command table starting from page 14 onwards)

And just feel free to make a WIP-PR and asks questions if you are stuck anywhere.

Alexx-G, 2021-04-02:
Hey @caemor @jamesmunns ,
Thanks for pointers. I'm still figuring it out. I have some weird issues when I try to test the 3.7 display using a NRF52 board. Things I've checked so far - initialization and display sequence match C++/Python examples, tried different LUT provided in those examples, width/height match as well, rotation only affects that tiny area where "circle" is shown.
Have you encountered this kind of effect?
[an attempt to draw a circle](https://user-images.githubusercontent.com/3869268/113447971-9628c880-9403-11eb-8a2d-e719fa8926ee.jpeg)

caemor, 2021-04-03:
@Alexx-G lets move to issue #68

CyanBlob, 2021-11-29:
Has anybody started working on the 7.5inch red/black/white yet? I've got a couple at home begging to be used for something! If nobody else has then I might take a look at adding support, but I'm pretty new to Rust so it may take me a while.

caemor, 2021-11-29:
Hello @CyanBlob,
I think @marcelbuesing has done some work on them here in https://github.com/caemor/epd-waveshare/pull/45, but you need to check if you have a v1 or v2 of the 7in5 display, since according to his messages there he should have used the v1 in the pr.

CyanBlob, 2021-11-29:
Mine actually have v3 stickers on them, which is odd since I haven't been able to find much about a v3 anywhere. Here's what I bought: https://www.aliexpress.com/item/1005001760018474.html?spm=a2g0s.9042311.0.0.79404c4dusNtBe   

There does seem to be a spec for the v3 (based on the URL): https://www.waveshare.com/w/upload/8/8c/7.5inch-e-paper-b-v3-specification.pdf  

The PDF itself just calls it B model though. These are my first e-ink displays, so I'm really not sure what the v3/B means. How different would you expect them to be from v1?

caemor, 2021-11-29:
I believe B/C is just the red/yellow difference mostly (D flexible, F "full" color,...). v1, v2, v3 are the same display size with sometimes better resolution,... and slightly different command tables.

So it looks like waveshare is currently slowly rolling out the v3s of the various display, but your doesn't seem to be implemented yet. 

So what I would recommend you is to use https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/c/lib/e-Paper/EPD_7in5b_V2.c of the v2, the specification you linked: https://www.waveshare.com/w/upload/8/8c/7.5inch-e-paper-b-v3-specification.pdf and marcelbuesings version next to each other.
First you could just update the command table. Panel setting is 0x00, panel setting 0x01 (r01h) and so on.
Then you change the init function as needed, the size! (better resolution), and then the other functions. Most likely you can just use quite a lot of it as it is already, but better be safe and compare all versions for these parts to be sure and then try it out.

If you want to go a safer route, wait until waveshare has released their own code for your v3.

CyanBlob, 2021-11-30:
Update! The display works! Kind of. It's currently kinda freaking out between white, black, red, and multicolor static right now when it *should* just be flipping back and forth between solid white and solid black. But I'm happy regardless. I ended up basing this off the 7in5_v2 version, so it doesn't handle the buffers properly (I think). I just wanted to get *something* happening because I never could before. Current work: https://github.com/CyanBlob/epd-waveshare/commit/77579f5bff12dbc9c40d83f9d62839c7679bf4c1 

I'm gonna keep working on this soon. Hopefully fixing it isn't that big of a deal!

Anshorei, 2022-01-14:
I've recently bought a Waveshare 7.5" Black/White/Red. It has a V3 sticker on the back.
- 7in5_v2: screen becomes patchy, but something vaguely looking like the expected output appears
- Cyanblob's 7in5_v3: screen flashes black/white/red in a pattern reminiscent of static noise (i.e. exactly as he describes in his post)

@CyanBlob any updates on your progress?

CyanBlob, 2022-01-15:
@Anshorei unfortunately I haven't had the time to figure out why my version is so broken, and I probably won't for quite a while. If you have the time I'd encourage you to take a look!

Anshorei, 2022-01-15:
@CyanBlob Thanks for the update. I do want to look into it, I've also got a 2in13bc V3 that I could try to get working, but I'm not sure how much time it'll take -- and of course how much time I'll be able to make for this.

Anshorei, 2022-01-17:
Worked on this a little and have a draft to support the 2in13b_v3 (#95 ) Everything I'm using in the example is working for me.

Anshorei, 2022-01-18:
Have not had time to look at the 7"5 display yet, but I ordered one of the 2"9 inch tricolor and of the 5"65 heptacolor displays from waveshare as well to try out.

simon0356, 2022-02-03:
Hi everyone, 
I have implemented support for the 4.2" module v2 3 color  (I use the following one : https://www.amazon.fr/gp/product/B075FRVC4L/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) 
I'm not trying to create for a pull request for now since i'm not a rust guru and i'm not 100% sure of my implementation.

I use the crate with the experimental port of rust on ESP32, i will try as soon as i get a new raspberry pi to implement the example  
I have already used the waveshare ESP32 Driver board with a Waveshare E-Paper 4.2" **without** PCB without applying any modification to the crate (this example is available here : https://github.com/ivmarkov/rust-esp32-std-demo) 
Since i broke my ESP32 Driver Board, i have bought a new 4.2" epaper **with** PCB wired to an AzDelivery ESP32.
This new E-Paper screen is labeled V2 and didn't work as is so i implemented an epd4in2_v2 crate with three color support but without implementing partial refresh (I don't known if it's realy possible). 
To do this i completely copy-paste the behaviour of the driver here : https://www.waveshare.com/wiki/File:E-Paper_ESP32_Driver_Board_Code.7z (which is not up to date) 

I'am waiting for a raspberry pi, and a new ESP32 Driver board to test further and maybe create a pull request 

Here is my fork : https://github.com/simon0356/epd-waveshare
[image]
[image]

Hope it could help anyone.
Simon

CyanBlob, 2022-05-18:
@Anshorei I just checked and the V2 Python sample code from here https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/examples/epd_7in5b_V2_test.py works for the V3 7in5. I'm not sure why the epd-waveshare V2 version isn't working but now that I have *something* working I can start looking into it more

CyanBlob, 2022-05-19:
[image]
Update, I have it mostly working. I copied the init process and the sleep timings from the Python project and then copied the TriColor display implementations from epd2in13bc.  
It mostly works, but the colors are coming out wrong. I’m gonna get that working and then do some cleanup and then I should be done!  
Doing this with my ESP32 has been fun but is definitely slowing things down since Rust support for it is still more or less in its infancy

simon0356, 2022-05-19:
I think i got the same issue, if I remenber the following line were important : 
https://github.com/simon0356/epd-waveshare/commit/7df315362acf71525efb30c7a4c48baa8afebbf5#diff-402c0d9cdd4ef7b8a30bf747d6556c063d0f763cf502a3602f10497397327963R98

"DisplayColorRendering::Positive"

marcelbuesing, 2022-05-19:
> [image] Update, I have it mostly working. I copied the init process and the sleep timings from the Python project and then copied the TriColor display implementations from epd2in13bc. It mostly [...]

The ESP32 integration is pretty cool, do you have some public repository to look at it? I am curious.

simon0356, 2022-05-19:
https://github.com/simon0356/rust-esp32-waveshare-4_2 with the v1  and waveshare esp32 board version out of the box

CyanBlob, 2022-05-19:
Here's my current work: https://github.com/caemor/epd-waveshare/compare/main...CyanBlob:main  

I can't get it to display white using 'clear_buffer'. I'm not sure what I'm missing. Can somebody familiar with three-color displays take a look? White is always displayed as black after using 'clear_buffer'. It seems to works fine when rendering text though. The docs are here: https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)  

I'm suspecting that the 7in5 V3 handles colors differently than other tricolor boards.  

My code is here: https://gitlab.com/CyanBlob/e-paper-rust.  

For those that want to try with an ESP32 themselves, you need to grab the toolchain from here: https://github.com/esp-rs/esp-idf-template. Some of my demo code came from here: https://github.com/ivmarkov/rust-esp32-std-demo

CyanBlob, 2022-05-21:
Opened PR: https://github.com/caemor/epd-waveshare/pull/109 

The problem was with how the 7in5 V3 handles colors. In the black/white buffer a 255 means white and 0 means black. But in the red/white buffer a 0 means white and a 255 means red, which is effectively inverted. That meant that when I did `display.clear_buffer(TriColor::White)`, the B/W buffer would be set to white but the R/W buffer would be set to red. Red has the highest priority, which messed up my colors.

arlyon, 2022-05-30:
@simon0356 great work, do you plan on opening a PR?

simon0356, 2022-06-14:
I'm currently try to make the screen work with my raspberry pi 1,  i will try to produce a PR with that

Tnze, 2022-07-31:
Hi, is [2.66inch e-Paper Module (B)](https://www.waveshare.net/wiki/2.66inch_e-Paper_Module_(B)) supported?

caemor, 2022-08-01:
> Hi, is [2.66inch e-Paper Module (B)](https://www.waveshare.net/wiki/2.66inch_e-Paper_Module_(B)) supported?

There is https://github.com/caemor/epd-waveshare/pull/107, which should have been version a. Version B isn't support yet.

piedoom, 2024-02-29:
I'm looking to support the 4.2in b/w/r v2.1 display. I'm a little confused at the A vs B vs C etc - what letter would this display be?

Are the steps still valid in 2024? I was wondering if I could update simon's previously mentioned branch and have that work, or if I have a different version of the module. 

Thanks!

simon0356, 2024-03-01:
Hi, 
I remember that it was not really clear how to distingue different version too, 
Feel free to try my verson, i didn't get time to produce a proper version to open a PR sorry.

Simon