rust-embedded-community/epd-waveshare #49: Error running 4.2" using an STM32F3

caemor, 2020-05-27:
> 
> 
> > Are there any tricks on the 4.2" using an STM32F3? Getting no signs of life from the display, and the program crashing once I hit the `let mut display = Display4in2::default();` line.
> 
> Then you most likely don't have enough ram left over. It needs 15kB for the full default display buffer and the Display4in2::default() doesn't do much besides making a 15kB framebuffer. You could look into the examples/epd4in2_variable_size.rs example which is using a smaller buffer to update only little parts of the display.
> 
> You could also check how much ram you have left and try to reduce your usage elsewhere.

Thank you. It also looks like this command: `    let mut epd =
        EPD4in2::new(&mut spi, cs_disp, busy, dc, rst, &mut delay).expect("eink initalize error");` is crashing it as well. I think I need to figure out how to set up panics to see what's going on.

_Originally posted by @David-OConnor in https://github.com/caemor/epd-waveshare/issues/47#issuecomment-634609753_

caemor, 2020-05-27:
@David-OConnor

How are you running it? 
Could you try running it with semihosting: https://docs.rust-embedded.org/book/start/semihosting.html so you could get the errors?

An old example using this library can be found here for the bluepill: https://github.com/caemor/epd-waveshare/tree/60f48a8273c43ddbe8ae840f895e4d88fb04be8a/examples/epd4in2_full_blue_pill

caemor, 2020-05-27:
Also like mgottschlag said, be aware that the current examples focus around using it on an embedded linux system (for simplification) so you need to change the spi and gpio stuff similar as it is done in the bluepill example above.

David-OConnor, 2020-05-27:
Thank you for creating this new issue. As far as I can tell, my setup code matches that from the example, minus changes to the clock and SPI setup that may be attributed to the F3 vs F1, or API versions, and I didn't have the `set_lut` step.

After further tests, panics are working, the `EPD4in2::new` step isn't causing a crash. It's the `display = Display4in2::default();` only, and it's failing/crashing with no error message. This is where my setup code differs from the example:

```rust
let clocks = rcc.cfgr.freeze(&mut flash.acr);

let sck = gpioa.pa5.into_af5(&mut gpioa.moder, &mut gpioa.afrl);
let miso = gpioa.pa6.into_af5(&mut gpioa.moder, &mut gpioa.afrl);
let mosi = gpioa.pa7.into_af5(&mut gpioa.moder, &mut gpioa.afrl);

```

David-OConnor, 2020-05-27:
I switched to the semihosting pattern in your link, and can confirm it works with panics and printing.

No error message. As soon as I add the line `let mut display = Display4in2::default();` to the code, the program refuses to show any signs of life, including `hprintln!` statements before it in the code, that work without that display line.

David-OConnor, 2020-05-27:
Looks like you were right about the memory. This has 48k. It's a barebones program right now with nothing else going on but trying to make this display work. This step alone: `let mut buffer = [epd4in2::DEFAULT_BACKGROUND_COLOR.get_byte_value(); 62500];` will cause a fail, although if the buffer size is lowered enough that step will pass, but the `display = VarDisplay...` step will still fail.

caemor, 2020-05-27:
This sounds pretty much like you hit a memory limit although it should only take 15k for the full display.

**For the size problem:**
- Can you try getting your size (https://docs.rust-embedded.org/faq.html#my-embedded-rust-program-is-too-big) ? (Longer description of this here: https://docs.rust-embedded.org/book/unsorted/speed-vs-size.html) 
- Could you maybe post a link to your code so I can look over it at the weekend (and try it if you are using stm32f3 discovery)? Also are you using https://github.com/stm32-rs/stm32f3xx-hal and maybe the rtfm framework or the new board support package (https://github.com/rubberduck203/stm32f3-discovery) and not the old one from japaric which is pretty outdated?

**For testing the display:**
Can you just try this really small example (128 Byte) to test the display?
```
let (x, y, width, height) = (0, 0, 32, 32);

let mut buffer = [0xFF; 128]; //32x32 Pixel /8 ==> 128 Byte
let mut display = VarDisplay::new(width, height, &mut buffer);
display.set_rotation(DisplayRotation::Rotate0);
draw_text(&mut display, "R", 0, 0);

display.set_rotation(DisplayRotation::Rotate90);
draw_text(&mut display, "R", 0, 0);

display.set_rotation(DisplayRotation::Rotate180);
draw_text(&mut display, "R", 0, 0);

display.set_rotation(DisplayRotation::Rotate270);
draw_text(&mut display, "R", 0, 0);

epd4in2
    .update_partial_frame(&mut spi, &display.buffer(), x, y, width, height)
    .unwrap();
epd4in2
    .display_frame(&mut spi)
    .expect("display frame new graphics");
```

caemor, 2020-05-27:
If this https://github.com/David-OConnor/rust-embedded-quickstart is the basis of the code you are using you might be missing the memory.x file

David-OConnor, 2020-05-27:
Thank you so much. I just discovered that my `openocd.gdb` file was causing the program to halt on HardFaults, which suppressed panics that happened after. Working through this, now with error messages, and will report with the results of your fixes, and where the panic rabit hole leads. Your suggestion to use panic semihosting significantly improved my workflow btw.

David-OConnor, 2020-05-27:
Oh my! Your example runs and updates the screen, and your initial intuition about a memory problem was correct. I think the problem I ran into based on the VarSize was a diff one, eg width/8 * height < len.

- Running into issues with `cargo size`; will post update once sorted.
- I'm indeed using the STM32F3Discovery, with the [first package](https://github.com/stm32-rs/stm32f3xx-hal) you linked.
- 

I've added a `memory.x` file to the root of my program, this this info gleaned from the Datasheet:
```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

David-OConnor, 2020-05-28:
Related Q: If you're diving the screen into multiple sections (Presumably this is how you relatively quickly refresh part of the page?), do you need to offset the first arg in the array by the memory used by the prev one? Ie:

```rust
    let mut buffer_top_left = [0xFF; (WIDTH * HEIGHT / 8) as usize];
    // In the line below, do we leave `0xFF`, or replace it with something higher, indicating
    // we've already used the `0xFF` slot? Getting a `assertion failed: width / 8 * height >= len'` error,
and trying to figure out if this is why.
    let mut buffer_bottom_left = [0xFF; (WIDTH * HEIGHT / 8) as usize];

    let mut disp_top_left = VarDisplay::new(WIDTH, HEIGHT, &mut buffer_top_left);
    let mut disp_bottom_left = VarDisplay::new(WIDTH, HEIGHT, &mut buffer_bottom_left);
```

caemor, 2020-05-28:
The VarDisplay is just an abstraction to use the embedded_graphics library with the buffer.

The buffer needs to be of type `[u8; (WIDTH * HEIGHT / 8) as usize]`;
So 0xFF is already the highest value in there (with setting 16 pixels to white)

> Can you use the same buffer for multiple sub-screens, to bypass the memory issue?

So yes but you shouldn't do it in the same way you do above as you are creating a new buffer for each Display.  A better way would be like below:

```Rust
let mut buffer = [0xFF; your_width * your_height / 8 ]; 
let mut display = VarDisplay::new(your_width, your_height, &mut buffer);

// Update first small part of the internal displaybuffer
display.set_rotation(DisplayRotation::Rotate0);
draw_text(&mut display, "R", 0, 0);

display.set_rotation(DisplayRotation::Rotate180);
draw_text(&mut display, "R", 0, 0);

epd4in2
    .update_partial_frame(&mut spi, &display.buffer(), first_x, first_y, your_height, your_height)
    .unwrap();

// Update second different part of the internal displaybuffer
display.clear_buffer(Color::White);
display.set_rotation(DisplayRotation::Rotate0);
draw_text(&mut display, "Different", 0, 0);

epd4in2
    .update_partial_frame(&mut spi, &display.buffer(), second_x, second_y, your_height, your_height)
    .unwrap();

// Update display from internal displaybuffer
epd4in2
    .display_frame(&mut spi)
    .expect("display frame new graphics");
```

It should also be possible to do it like this:
```Rust
   
{
    let mut buffer_left = [0xFF; (30 * 200 / 8) as usize]; //32x32 Pixel /8 ==> 128 Byte

    let mut disp_left = VarDisplay::new(30, 200, &mut buffer_left);

    // do some stuff with this display

    // and then update the internal framebuffer
    epd4in2
    .update_partial_frame(&mut spi, &display.buffer(), second_x, second_y, your_height, your_height)
    .unwrap();
} // now the memory of buffer_left should be freed afaik

{
    let mut buffer_bottom_left = [0xFF /*White*/; (WIDTH * HEIGHT / 8) as usize];

    
    let mut disp_bottom_left = VarDisplay::new(WIDTH, HEIGHT, &mut buffer_bottom_left);

    // do some stuff with this display

    // and then update the internal framebuffer
    epd4in2
    .update_partial_frame(&mut spi, &display.buffer(), second_x, second_y, your_height, your_height)
    .unwrap();
} // now the memory of buffer_bottom_left should be freed afaik

// Update display from internal displaybuffer
epd4in2
    .display_frame(&mut spi)
    .expect("display frame new graphics");
```

But it also should be possible to use the full 15k Display buffer with your 40k RAM. Not sure where you loose all the space

See the documentation for more stuff: https://docs.rs/epd-waveshare/0.4.0/epd_waveshare/graphics/struct.VarDisplay.html (especially the Display trait implementation)

David-OConnor, 2020-05-28:
Thank you very much for the description, and detailed examples - I now have a better understanding of the buffer, and how it interacts with the display. That code is working as you describe, and the Display trait docs are helpful.

David-OConnor, 2020-05-28:
As you implied, creating a `memory.x` solved the memory / `Display4in2::default()` issue! 

Contents (STM32F303 Discovery):
```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

caemor, 2020-05-28:
> Would it help if I PRed a STM32F3 example once I work through this more?

I think it would be better to add link to a working example in your repo, since I can't support all the examples for each device in here.

> Is there a recommended way of updating the Display/buffer on only a specific region, eg to erase something, or swap out a small piece of text or graphic without refreshing the whole screen? Ie something that would be described as display_partial_frame(). Or is this in the area of set_lut()?

Yes the refresh is defined through the sram data + LUT. So you can improve the time by doing small updates with `update_partial_frame` and use "dangerous" LUTs with faster refresh time (and no full b+w+b+w refreshs). Speeding up the clock to near its upper limit from the datasheet also helps.

The Display Refresh only does the following according to its description (and we only send a single command for this):
```Rust
    /// While user sent this command, driver will refresh display (data/VCOM) according to SRAM data and LUT.
    ///
    /// After Display Refresh command, BUSY_N signal will become “0” and the refreshing of panel starts.
    DISPLAY_REFRESH = 0x12,
```

Here are some old timing I took how long different stuff takes on this epd:
[image]

David-OConnor, 2020-05-30:
I really appreciate all the help!

caemor, 2021-01-27:
Is this resolved?

mydogisbox, 2021-09-06:
@David-OConnor if you have an example I would find that very helpful! Do you have a link to a repo or some such?

David-OConnor, 2021-09-06:
I don't have any open source repos using this, but I'll say confidently that this library is the best generic Rust embedded driver I've used, and its official examples are quite good. What are you having problems with in particular?

David-OConnor, 2021-09-06:
> 
> 
> Is this resolved?

Yep!

mydogisbox, 2021-09-06:
I'm trying to get the basic setup to compile but I'm very new to embedded rust and am not sure how to translate this old example:
https://github.com/caemor/epd-waveshare/blob/60f48a8273c43ddbe8ae840f895e4d88fb04be8a/examples/epd4in2_full_blue_pill/src/main.rs

I'm specifically stuck on what `into_alternate_push_pull` should be replaced with:
```
    let sck = gpiob.pb13.into_alternate_push_pull(&mut gpiob.crh);
    let miso = gpiob.pb14;
    let mosi = gpiob.pb15.into_alternate_push_pull(&mut gpiob.crh);
    let mut spi = spi::Spi::spi2(
        device.SPI2,
        (sck, miso, mosi),
        epd_waveshare::SPI_MODE,
        4.mhz(),
        clocks,
        &mut rcc.apb1,
    );
```

mydogisbox, 2021-09-06:
I can open a new issue if you'd prefer to close this one @caemor

David-OConnor, 2021-09-06:
That's an API specific to the HAL you're using. I recommend starting at the examples there. Which one is it?

mydogisbox, 2021-09-06:
I'm also using the STM32F3 discovery board.

David-OConnor, 2021-09-06:
See [this example](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/examples/spi.rs#L33) on STM32f3xx-hal.

The code I use for the 4.2" EPD on a different HAL:
```rust
let _sck = Pin::new(Port::A, 5, PinMode::Alt(5));
let _miso = Pin::new(Port::A, 6, PinMode::Alt(5));
let _mosi = Pin::new(Port::A, 7, PinMode::Alt(5));

// ...

    let spi_cfg = SpiConfig {
        // The EPD appears to work on all but the following modes: IdleLow, SecondTransition.
        // Mode 3 meets this requirement.
        mode: SpiMode::mode3(),
        ..Default::default()
    };

let mut spi = Spi::new(
    dp.SPI1,
    SpiDevice::One,
    spi_cfg,
    // With an 80Mhz SPI periph speed, the speeds near this range are 80/16 = 5Mhz.
    BaudRate::Div32,
    );

```

mydogisbox, 2021-09-06:
Ah, that example is helpful. I'll play around with it for a bit. Thanks!

caemor, 2021-09-08:
@David-OConnor Thanks for helping out! :smile: 

@mydogisbox Since your issue matches this issues topic we can stay reuse this issue for you. Have your issues already been resolved with Davids help?

mydogisbox, 2021-09-10:
I'm pretty sure I'm doing something wrong here:
```
fn main() -> ! {
    let core = cortex_m::Peripherals::take().unwrap();
    let device = stm32f3::stm32f303::Peripherals::take().unwrap();
    let mut rcc = device.RCC.constrain();
    let mut flash = device.FLASH.constrain();

    let clocks = rcc
        .cfgr
        .use_hse(Megahertz(8))
        .sysclk(Megahertz(72))
        .pclk1(Megahertz(36))
        .freeze(&mut flash.acr);

    let mut gpioa = device.GPIOA.split(&mut rcc.ahb);
    let mut gpiob = device.GPIOB.split(&mut rcc.ahb);
    let mut gpioc = device.GPIOC.split(&mut rcc.ahb);

    let mut delay = delay::Delay::new(core.SYST, clocks);

    let sck = gpioc
        .pc10
        .into_af6_push_pull(&mut gpioc.moder, &mut gpioc.otyper, &mut gpioc.afrh);
    let miso = gpioc
        .pc11
        .into_af6_push_pull(&mut gpioc.moder, &mut gpioc.otyper, &mut gpioc.afrh);
    let mosi = gpioc
        .pc12
        .into_af6_push_pull(&mut gpioc.moder, &mut gpioc.otyper, &mut gpioc.afrh);

    let mut spi: stm32f3_discovery::stm32f3xx_hal::spi::Spi<
        stm32f3::stm32f303::SPI3,
        (
            stm32f3_discovery::stm32f3xx_hal::gpio::Pin<
                stm32f3_discovery::stm32f3xx_hal::gpio::Gpioc,
                stm32f3_discovery::stm32f3xx_hal::gpio::U<10_u8>,
                stm32f3_discovery::stm32f3xx_hal::gpio::Alternate<
                    stm32f3_discovery::stm32f3xx_hal::gpio::PushPull,
                    6_u8,
                >,
            >,
            stm32f3_discovery::stm32f3xx_hal::gpio::Pin<
                stm32f3_discovery::stm32f3xx_hal::gpio::Gpioc,
                stm32f3_discovery::stm32f3xx_hal::gpio::U<11_u8>,
                stm32f3_discovery::stm32f3xx_hal::gpio::Alternate<
                    stm32f3_discovery::stm32f3xx_hal::gpio::PushPull,
                    6_u8,
                >,
            >,
            stm32f3_discovery::stm32f3xx_hal::gpio::Pin<
                stm32f3_discovery::stm32f3xx_hal::gpio::Gpioc,
                stm32f3_discovery::stm32f3xx_hal::gpio::U<12_u8>,
                stm32f3_discovery::stm32f3xx_hal::gpio::Alternate<
                    stm32f3_discovery::stm32f3xx_hal::gpio::PushPull,
                    6_u8,
                >,
            >,
        ),
        u8,
    > = spi::Spi::spi3(
        device.SPI3,
        (sck, miso, mosi),
        epd_waveshare::SPI_MODE,
        Hertz(3_000_000),
        clocks,
        &mut rcc.apb1,
    );

    let mut epd4in2 = epd_waveshare::epd4in2::EPD4in2::new(
        &mut spi,
        gpiob.pb12.into_push_pull_output(&mut gpiob.crh),
        gpioa.pa10.into_floating_input(&mut gpioa.crh),
        gpioa.pa8.into_push_pull_output(&mut gpioa.crh),
        gpioa.pa9.into_push_pull_output(&mut gpioa.crh),
        &mut delay,
    );

    loop {}
}
```

That last bit doesn't compile since those functions don't even take that number of parameters anymore.

```
    let mut epd4in2 = epd_waveshare::epd4in2::EPD4in2::new(
        &mut spi,
        gpiob.pb12.into_push_pull_output(&mut gpiob.crh),
        gpioa.pa10.into_floating_input(&mut gpioa.crh),
        gpioa.pa8.into_push_pull_output(&mut gpioa.crh),
        gpioa.pa9.into_push_pull_output(&mut gpioa.crh),
        &mut delay,
    );
```

And uh... obviously if there is a way to eliminate the need for that type annotation on the spi definition that would be substantially better.

I feel like I ought to be able to educate myself enough to figure out some of these answers, but I'm not quite sure where to look.  Let me know if there is a better way to find these answers then just asking every time I can't get it to compile.

mydogisbox, 2021-09-19:
I found the information I needed to make progress and got it to compile. I have a 2.9" (d) e-paper module so I'll need to add support for that, but I'll open a new thread for that. Thanks!

caemor, 2021-09-25:
Thats great to hear :-)

closed 2021-09-25 by caemor