rust-embedded-community/epd-waveshare #50: Dividing the dims by 8

David-OConnor, 2020-06-12:
Question: What is the purpose of dividing the dims by 8 when creating the buffer? eg:
```rust
impl Default for Display4in2 {
    fn default() -> Self {
        Display4in2 {
            buffer: [DEFAULT_BACKGROUND_COLOR.get_byte_value();
                WIDTH as usize * HEIGHT as usize / 8],
            rotation: DisplayRotation::default(),
        }
    }
}
```
With this in mind, how should you map pixels to the buffer, given it's not a 1:1 match? Thank you.

caemor, 2020-06-29:
Each pixel only has 1 bit of info about it stored (Is it black or white). So we can put 8 Pixel in one Byte/u8 and therefore reduce storage.

David-OConnor, 2020-06-29:
Thank you. That answers my question.

closed 2020-06-29 by David-OConnor