rust-embedded-community/epd-waveshare #57: Support for EPD 2in13(D)

hugolm84, 2020-11-05:
Hi! First and foremost, thanks for all the work that been put in to this project!

I have an EPD 2in13 D e-paper, (Datasheet: https://www.waveshare.com/w/upload/5/5b/2.13inch_e-Paper_%28D%29_Datasheet.pdf , Commands(?): https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/c/lib/e-Paper/EPD_2in13d.c), and have issues figuring out how to add support for it.

Looking in issue #26 I learned to look at the datasheet to figure out the Commands. However, It's a bit confusing as the example 2in13v2 (https://www.waveshare.com/w/upload/e/e6/2.13inch_e-Paper_Datasheet.pdf) shows the commands with hex values, but it seems omitted for 2in13d? 
Do you have any pointers as how to figure out the values?

Thanks!

dkm, 2020-11-08:
Hi,
Not sure exactly what your problem is. For 2in13v2 B I've linked the doc PDF here : https://github.com/caemor/epd-waveshare/blob/main/src/epd2in13_v2/mod.rs
I've also tried to make sense out of the hexa values when possible. Maybe this can help ? The most useful documentation to understand hardcoded values in the examples in my case was this one : http://www.e-paper-display.com/download_detail/downloadsId=682.html

caemor, 2020-11-12:
Thanks but its the work of many others like you who add display after display :partying_face: 

You can find the the commands in your datasheet starting from page 8 `6.Command table`:
[image]

The hex value of the command is e.g. for the power setting 01h as its written in blue, the other blue numbers before are the single bits that lead to this hex value.
Afterwards you need to send 5 data bytes for power setting with the default value to be send also shown on the right. 

You can see the same happening here as well: https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/c/lib/e-Paper/EPD_2in13d.c#L319 
In this example you can also see that the datasheet differs slightly from the waveshare implementation (slightly different voltage chosen 0x26 vs 0x2b). So sometimes you need to decide which one makes more sense, test it and go with it (and maybe add a small note/comment there as well)

Feel free to ask more questions whenever you need help, you can also an early merge request or send a link to your progress here so others and I can take a look into it.

hugolm84, 2020-11-12:
@caemor Hi! Thanks for this great explanation, I  was a bit thrown off by the  0x26 -> 0x2b and did not connect the dots. 

I will try to add support for it and create a WIP PR when I can. Thanks again!

caemor, 2021-01-27:
Hello Hugo, is there anything I can help you with?

hugolm84, 2021-01-28:
Hi! Time flies! I did not have time for this!

closed 2021-01-28 by hugolm84

jaxter184, 2023-04-09:
This isn't done, right? Can it be reopened? I've got a [barely working fork](https://github.com/jaxter184/epd-waveshare) that I can PR in like a few weeks (assuming I get around to cleaning it up and testing it more rigorously).

caemor, 2023-04-09:
No, this isn't done. I can either reopen this issue or you create a new one.

jaxter184, 2023-04-09:
Whichever option is more consistent with how the project is currently organized is fine with me, I don't really have a preference either way. If there isn't an open issue by the time I get my fork cleaned up, I'll go ahead and make a new issue.