rust-embedded-community/epd-waveshare #70: 7.5" V2 display driver does not work

olback, 2021-04-04:
A few things I found while trying to get my display to work as expected:
* `epd7in5_v2v::EPD7in5::new(...)` hangs [here](https://github.com/caemor/epd-waveshare/blob/main/src/epd7in5_v2/mod.rs#L72) since the `wait_until_idle` implementation is wrong. In the [example C code provided by Waveshare](https://github.com/waveshare/e-Paper/blob/7c368a22afb5456fb4622c617e792bd2f83e878b/RaspberryPi_JetsonNano/c/lib/e-Paper/EPD_7in5_V2.c#L78) one can see that the driver sends a `GET_STATUS` command and waits 20ms.  This seems to be needed otherwise the busy pin wont update which is the root cause to the problem.

* Black and White seems to have been mixed up. Setting the fill-color to black makes the display white and vice versa. Same thing goes for text colors/background.

**A few things that might be important:**
* Cross-compiling from Arch Linux x64_64 to aarch64 with `aarch64-linux-gnu-gcc` targeting `aarch64-unknown-linux-musl`.
* Using the [rppal](https://github.com/golemparts/rppal) crate with the hal-unproven feature.
* Running on Raspberry Pi OS 64-bit Beta (Kernel 5.4.51-v8+) on a Raspberry Pi 3b+.

A hacked together version of epd-waveshare can be found [here](https://github.com/olback/epd-waveshare/tree/7.5hack) and an example can be found [here](https://github.com/olback/smart-home/blob/epd-working/portal/src/main.rs).

caemor, 2021-04-18:
Merged #71

caemor, 2021-04-18:
What is still missing from here:
- black/white stuff  /cc @whiite 
- Optionally a testcase for the delay

Flocksserver, 2021-11-30:
Hey :) 
I am facing the same problem. My proof of concept hangs at `epd7in5_v2v::EPD7in5::new`. I am not so familiar with the hardware related programming and the c code provided by waveshare. Is there any plan to fix this? Or do we have to find a proper fix ourselves and create a PR? 
Thanks for your work 💪

caemor, 2021-11-30:
Hello Flocksserver,
the delay issue was already fixed and merged in #71 
So your issue is most likely slightly different. Which version of this crate are you using? There was quite a long break of crates.io releases from my end until this week, so you might still be stuck on the old, unpatched version.
Another possible failure could be your display. Are you sure you have a v2 or maybe you got the newest v3 that they are slowly rolling out for the various displays? Or maybe we got a regression or bug somewhere else

Flocksserver, 2021-11-30:
Wow your second name is "The Flash". That was a fast reply :) 
I got it running, but I don't know what exactly the issue was. Probably not in this crate. 
The change was, that now I am using the rppal dependency (olback mentioned this crate) for initializing spi, pins etc. I use an old RPI 1 Model B with armv6. Maybe there is an issue in linux_embedded_hal for that architecture, who knows. Not your problem 😎 🥳

whiite, 2022-11-16:
Got a chance recently to take a look at this and I believe the culprit might be the change made in #83 which writes bytes one at time instead of all at once as reverting this got my 7.5" HD display working again and suspect this may be the case for the V2 too.
Happy to make a PR to sort this one out but seems like we may need to support both depending on the display, don't suppose we know which screens will need each write method?

GBouerat, 2023-01-07:
Hi,

I just bought a "7.5 Inch B/W/R V2/V3" with the e-Paper Driver HAT https://www.waveshare.com/7.5inch-e-Paper-HAT-B.htm

And after a day, I didn't succeed to display something on the screen with a raspberry 4 and a raspberry 3 😞
(the c demo from the documentation works https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B)_Manual#C_2)

Can someone help me or provide a working sample ? please,

Thanks

whiite, 2023-01-07:
Hey, out of curiosity does this fork work for you instead: https://github.com/whiite/epd-waveshare
This includes the change I talked about above that was needed for my 7.5” HD to work

caemor, 2023-01-07:
@whiite If you have time, could you test #141 (branch fix_single_write_for_7in5)? I reverted the change of #83 for all 7in5 displays by adding a const switch case.

whiite, 2023-01-08:
@caemor yeah sure thing, will probably be next week that I get the chance and I can only check for the HD version at the moment. Looks like a neat solution!

GBouerat, 2023-01-08:
> @GBouerat out of curiosity does this fork work for you instead: https://github.com/whiite/epd-waveshare This includes the change I talked about above that was needed for my 7.5” HD to work

I try everything I found include this fork, and nothing works for me so far, that's why I ask for some help 👍

caemor, 2024-12-04:
Since I hope this got solved by #141 I'm going to close this for now. If it is still an issue, feel free to either write a comment or open a new issue with for the specific issue.

commit 9a1575b2 (2021-04-18): API change to partially fix #70
closed 2024-12-04 by caemor