rust-embedded-community/epd-waveshare #90: Need help running the example for 2in13_v2

ole-treichel, 2021-11-25:
Hey, thanks for the library!

I am trying to run the example in `examples/epd2in13_v2.rs` on a raspberry pi 4. The display is directly connected to the pi via the displays header.

I verified the display is working with the official c example from [here](https://www.waveshare.com/wiki/Template:Raspberry_Pi_Guides_for_SPI_e-Paper).

I cloned the repo and tried two ways of running the example via `sudo`:

`sudo env "PATH=$PATH" cargo run --example epd2in13_v2` (from a normal pi user session)

and directly as root with

`cargo run --example epd2in13_v2`.

However nothing happens, no output is printed to the terminal, and the display stays blank.

Is it possible that the pins are not identical to the "default" pin configuration of the header?

I am really stuck here, any help on how to debug the issue would be much appreciated!

lrbalt, 2021-11-25:
Did you try running as root (sudo)? Otherwise you can try the pins from 2in13bc. I needed to change the default to those in that example for my v1 to work. I do not have a V2 though to test

ole-treichel, 2021-11-25:
Awesome that worked! Thanks for the super fast reply 👍

closed 2021-11-25 by ole-treichel