rust-embedded-community/epd-waveshare #113: Delay is too large when hardware reset.

Tnze, 2022-08-02:
According to [this specification](https://www.waveshare.net/w/upload/e/ec/2.66inch-e-paper-b-specification.pdf#13.2), delay between RESE low and RESE high should be 200us. But the `DELAY` is a `DelayMs<u8>`.

caemor, 2022-09-29:
We need to change delay to DelayUS<u32> to be able to use the `us` range as well as up to ~100ms

caemor, 2022-11-28:
/cc @peckpeck Do you think it might make sense to change DelayMS to DelayUS everywhere or is up to 100ms not enough for your usecase?

peckpeck, 2022-11-29:
The way forward is `DelayUs<32>` because it's the only trait supported in embedded-hal 1.0

peckpeck, 2022-12-03:
https://github.com/caemor/epd-waveshare/pull/131

caemor, 2022-12-03:
Hello @Tnze,
this should be resolved by Peckpecks PR in #131 where DelayMs<u8> was replaced by DelayUs<u32>

closed 2022-12-03 by caemor