rust-embedded-community/epd-waveshare #121: Could code be more generic ?

peckpeck, 2022-10-31:
Firt let me point that most waveshare epd are identical as goodisplay products.
Goodisplay better document their product and include the integrated circuit name in de datasheet.
This way we can know which epd has the exact same IC as which one.

In this case, I think we should just write generic driver for those ICs and instanciate them with some specific values for each epd we want to support. This would greatly reduce code size and make bug fix propagate to similar epd that share the same code.

For example 7in5b shares the exact same datasheet as 5in83b and as UC8179c IC at goodisplay. Only some data change.
I'd like to make a 
```
// generic impl
struct UC8179c<S: UC8179cSettings> { ... }
impl<S> UC8179c<S> { ... }

// specific EPD
impl UC8179cSettings for Epd7in5bSettings { ... }
type Epd7in5b = UC8179c<Epd7in5bSettings>;
```

where the `Setting` would just be a bunch of constants

This would also simplify extracting default LUT data (which format is common to a given IC) to play with them

peckpeck, 2022-11-01:
Ok I withdraw that one, exact integrated circuits are not easy to find, and the benefit in term of code lines is not high ven if it would hep a little.

peckpeck, 2022-11-01:
However a big benefit we could get is by making the graphics part more generic they are almost all the same, all that change are the dimensions and the color type.

peckpeck, 2022-11-03:
Here is how I see it https://github.com/caemor/epd-waveshare/pull/123

caemor, 2022-11-04:
I haven't found time to look at your code yet, but yes the library needs a bit of a rewrite to make it more generic.

Also I am planning to rename the crate in the near feature to make it more generic since as you said most waveshare displays are from gooddisplay anyway.

peckpeck, 2022-11-29:
Fixed by https://github.com/caemor/epd-waveshare/pull/123

closed 2022-11-29 by peckpeck