rust-embedded-community/epd-waveshare #122: The WaveshareDisplay associated color is not properly documented

peckpeck, 2022-11-01:
Since it is an associated type, most driver implement it as the color type the epd supports.

However since there is a separate WaveshareThreeColorDisplay trait, some driver implement it only as a black and white color and let the 3 colors handling to WaveshareThreeColorDisplay (eg epd1in54c).

We should document and enforce the right behaviour to make sure this is coherent between various implementations.

caemor, 2022-11-04:
Sounds like a good idea :+1:

peckpeck, 2022-11-23:
Any news on this ?

caemor, 2022-11-28:
```
DisplayColor should be the exact color type supported by the device. 

Please implement WaveshareThreeColorDisplay`, if you want to allow that in color mode the user can sometimes only update chromatic or B/W buffers separately to maybe gain some refresh time.

Please implement `WaveshareDisplay` with the bicolor type "Color" on a bicolor version of your structure, if you want to support a pure black and white mode on a color EPD, as this can speed up the refresh.

If there is no faster way, the user can still use a TriColor `WaveshareDisplay` as Bicolor one.
```

Some small changes but the meaning hopefully should have stayed the same. 

This also sounds like a good change to make the usage and development of this crate more intuitive.