rust-embedded-community/epd-waveshare #124: Use the same color representation for all EPD

peckpeck, 2022-11-06:
The user has to know how EPD implements colors because tricolors have split buffers and the user has to handle it if she doesn't use Display structures.

Moreover Display structures also have to know this too (see the hacky bwrbit in the structure, or the DisplayColorRendering).

I think it should be the only responsibility of epd structures to handle this, we should present a common simple buffer format ti the user for all epds and the specific epd structures should adapt the bits they send to their specific case.

caemor, 2022-11-28:
:+1: Combined with the color definition update this would be great usability update.

peckpeck, 2022-12-06:
I'm not sure anymore bout this.
This only impacts users of tricolor epd, and it means having to split data buffers which can be compute intensive.
Moreover, almost only people disabling embedded-graphics could be interested in this, and they are the most likely to be cpu constrained. 

So i'm starting to consider leaving the epd methods mostly unchanged, but instead move all color hacks into the colors structures so that the `Display` structure can abstract them.