rust-embedded-community/epd-waveshare #130: Issues trying to draw embedded graphics image

mangelajo, 2022-12-01:
163 |             let bus = Image::new(&ImageRaw::new(include_bytes!("../../icons/Bus.raw"), 30),
    |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `From<()>` is not implemented for `epd_waveshare::color::Color`
    |

I guess it's missing some conversions?

It's defined as 
```
impl<'a, C, BO> ImageRaw<'a, C, BO>
where
    C: PixelColor + From<<C as PixelColor>::Raw>,
    BO: ByteOrder,
{
```

I also tried creating a BinaryColor raw image, but then I cannot draw it on a "Color" display.

would it be possible to standardize into "BinaryColor" from embedded_graphics?, I could be missing lots of things, ... still new to rust.

mangelajo, 2022-12-01:
Hmm, I am getting it to work by adding this to color.rs:

```
#[cfg(feature = "graphics")]
impl From<BinaryColor> for Color {
    fn from(b: BinaryColor) -> Color {
        match b {
            BinaryColor::On => Color::Black,
            BinaryColor::Off => Color::White,
        }
    }
}
```

and then adding .color_converted() to the display:

```

    bus.draw(&mut display.translated(Point::new(100,100)).color_converted()).unwrap();
     
```

peckpeck, 2022-12-03:
yes some traits are missing to work with every color

peckpeck, 2022-12-05:
Fixed in https://github.com/caemor/epd-waveshare/pull/132

PR #132 (2022-12-07): Add embedded-graphics traits for color
Closes #130
closed 2022-12-07 by caemor