rust-embedded-community/epd-waveshare #149: Epd7in5: display.clear(TriColor::White) throws index out of bounds

GBouerat, 2023-03-24:
With `epd-waveshare = { git = "https://github.com/caemor/epd-waveshare" }` in Cargo.toml

This call to clear() crash with an `index out of bounds` exception

```
    let mut epd7in5 = Epd7in5::new(&mut spi, cs, busy, dc, rst, &mut delay, None)
        .expect("error");

    let mut display = Display7in5::default();

    display.clear(TriColor::White).ok();
    ```

https://github.com/caemor/epd-waveshare/blob/f98c12160fe211d2cc83188419a9ec4e855c789d/src/graphics.rs#L353

```
thread 'main' panicked at 'index out of bounds: the len is 48000 but the index is 48000', /cargo/git/checkouts/epd-waveshare-201024d8b8447899/f98c121/src/graphics.rs:353:25
stack backtrace:
   0: rust_begin_unwind
             at /rustc/8460ca823e8367a30dda430efda790588b8c84d3/library/std/src/panicking.rs:575:5
   1: core::panicking::panic_fmt
             at /rustc/8460ca823e8367a30dda430efda790588b8c84d3/library/core/src/panicking.rs:64:14
   2: core::panicking::panic_bounds_check
             at /rustc/8460ca823e8367a30dda430efda790588b8c84d3/library/core/src/panicking.rs:159:5
   3: epd_waveshare::graphics::set_pixel
             at /cargo/git/checkouts/epd-waveshare-201024d8b8447899/f98c121/src/graphics.rs:353:25
   4: epd_waveshare::graphics::Display<_,_,_,_,COLOR>::set_pixel
             at /cargo/git/checkouts/epd-waveshare-201024d8b8447899/f98c121/src/graphics.rs:161:9
   5: <epd_waveshare::graphics::Display<_,_,_,_,COLOR> as embedded_graphics_core::draw_target::DrawTarget>::draw_iter
             at /cargo/git/checkouts/epd-waveshare-201024d8b8447899/f98c121/src/graphics.rs:110:13
   6: embedded_graphics_core::draw_target::DrawTarget::fill_contiguous
             at /cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-core-0.3.3/src/draw_target/mod.rs:392:9
   7: embedded_graphics_core::draw_target::DrawTarget::fill_solid
             at /cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-core-0.3.3/src/draw_target/mod.rs:408:9
   8: embedded_graphics_core::draw_target::DrawTarget::clear
             at /cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-core-0.3.3/src/draw_target/mod.rs:423:9
             ```

vhdirk, 2023-11-07:
I get this on the Epd7in5 (v1), v2, bv2 and v3, too. @GBouerat have you found any workaround in the meantime?

GBouerat, 2023-11-07:
> I get this on the Epd7in5 (v1), v2, bv2 and v3, too. @GBouerat have you found any workaround in the meantime?

No 😞

vhdirk, 2023-11-07:
Although I haven't gotten the display itself to work, https://github.com/caemor/epd-waveshare/pull/177 fixes this bug.
It should already be fixed for 7in5b_v2, too. The other 7.5" displays are just B/W, so this does not apply.

PR #177 (2024-01-11): Remove epd7in5_v3 in favour of edp7in5b_v2
Fixes #149 .

These displays are exactly the same. 7in5_v3 did not work so there's no reason to keep it
closed 2024-01-11 by caemor