rust-embedded-community/epd-waveshare #161: Some 1.54" display links in README 404.

shymega, 2023-06-11:
Hi,

I'm trying to work out if my e-ink display is supported by this library. I was looking at the 1.54" display links and two 404 for me. I've made a PR since to organize the list, so this issue might be outdated if that's merged, but the following links are 404'ing.

1.54 Inch B/W/Y (C): https://www.waveshare.com/1.54inch-e-paper-c.htm
1.54 Inch B/W/R (B): https://www.waveshare.com/product/modules/oleds-lcds/e-paper/1.54inch-e-paper-module-b.htm

The other 1.54" display link is fine.

Is there a link we should be using instead of these broken links?

Cheers.

caemor, 2023-08-30:
Resolved by #160

closed 2023-08-30 by caemor

shymega, 2023-08-30:
@caemor #160 didn't resolve this error - it was a tidying-up commit, but two of the 1.54" displays are still 404'ing.

Can we re-open this issue?

caemor, 2023-08-30:
Sure

shymega, 2023-08-30:
Thanks. I don't know if those 404'ing links have a model number, but should be easy to find - I imagine?

shymega, 2023-08-30:
OK, so I've found the 'B' display, but 'C' is missing.

I think the best approach is to nullify the URI for the 1.54 ('C') display and update the 'B' display's link. We can add a note to the 'C' display, and say it's no longer available; which is what it would seem.

I can submit a PR now if that's a good plan?

caemor, 2023-08-30:
Sounds good to me :+1:

shymega, 2023-08-30:
Do you think it's worth contacting Waveshare to enquire if the link will come back? I hope it will. I don't mind submitting an email and forwarding their response to you if you would like me to.

shymega, 2023-08-30:
PR created.

caemor, 2023-09-01:
> Do you think it's worth contacting Waveshare to enquire if the link will come back? I hope it will. I don't mind submitting an email and forwarding their response to you if you would like me to.

You can try if you want. But most likely they are just not selling that display anymore.

shymega, 2023-09-05:
I've had a response.

In terms of README changes, they've given me a new URL, but it has also been discontinued. For reference purposes, I think it's best I open a new PR correcting the link.

commit 6948a631 (2023-08-31): Update 'B' 1.54" e-ink display, and remove 'C' display link

The 1.54" B/W/R (B) e-ink display URI has changed. However, the 1.54"
B/W/Y (C) e-ink display is no longer available.

Resolves #161.