rust-embedded-community/epd-waveshare #163: Example 2in13bc gets stuck on `Epd2in13bc::new()`

teenangst, 2023-07-14:
Managed to get it running after various permission errors, I am running the example as sudo (building not sudo, then running sudo) however nothing happened, I added in printlines between every line and the program gets stuck on `Epd2in13bc::new()`, there are no errors and no progress.

I saw there was issue #90 which isn't the same as they wanted 2in13_v2 for black and white while mine is 2in13bc which is black white and red. I did try out the 2in13_v2 example and no output but it did finish with "Finished tests - going to sleep" which is not the same as them.

SPI is enabled. I assume it goes without saying but just to be safe, I am using a Raspberry Pi (4B)

norbertkeri, 2023-09-21:
Are you wiring it up to the raspberry pi4 yourself? I /think/ I had this problem with an esp32c when my wiring and/or the code using the gpio pins were wrong.

abergmeier, 2024-03-03:
> Are you wiring it up to the raspberry pi4 yourself? I /think/ I had this problem with an esp32c when my wiring and/or the code using the gpio pins were wrong.

How does one then know which pins to use?

caemor, 2024-10-28:
If there aren't any permission issues, I also believe that it might be the wiring. Did you try some of the python examples from waveshare?

I am happy to open this issue once more, when feedback arrives.

closed 2024-10-28 by caemor