rust-embedded-community/epd-waveshare #164: Support for 12.48inch module

barafael, 2023-07-30:
Is support for [this waveshare 12.48 inch module](https://www.waveshare.com/12.48inch-e-paper-module-b.htm) feasible, or planned?

caemor, 2023-08-30:
It should be feasible, but it's not something I've planned as I don't have one of these in my hands.

barafael, 2023-08-31:
I have one. It works fine with the C example code from waveshare, but yeah that's not great. I can either help test, or, if you like, I can send it to you. No obligations other than sending it back to me when you don't want it anymore, and only if it didn't break along the way :D let me know what you think, email is on my profile. I can send to a Packstation ofc.

barafael, 2023-09-01:
OK. I'll let you know when I need help, could be a while. Thanks!

closed 2023-09-01 by barafael

vadimcn, 2023-10-01:
Hi!  
I have the display and I wanted to take a stab at implementing the diver.   

However... I immediately ran into conceptual difficulties with this crate's traits, because they [assume that all displays have the same interface](https://github.com/caemor/epd-waveshare/blob/main/src/interface.rs#L5), and take a fixed number of pins as generic parameters.   Well, as it turns out, epd12in48(B) actually consists of [4 sub-displays](https://www.waveshare.com/wiki/12.48inch_e-Paper_Module), each with its own CS and BUSY pins, while DC and RST pins are by 2 sub-displays).  

@caemor, would you have any advice as to how to fit this sort of device into your framework?  

The only thing that comes to mind, is exposing it as 4 separate displays.  Which might have to be done anyways, since my ESP32 driver board [apparently cannot allocate the contiguous 160K bufer](https://www.esp32.com/viewtopic.php?f=19&p=116093), that would be needed for a 1304*984 image.