rust-embedded-community/epd-waveshare #165: MH-ET LIVE epaper support

oyvindnetland, 2023-08-24:
Hi, I just got a cheap epaper device of the brand MH-ET LIVE. Thought they where supposed to be compatible with waveshare, but apparently not. I got a 2.9 inch white/red/black display, and none of the 2.9 examples here worked. I managed to get something kinda working with 2.15b V4 example from waveshare python library, but resolution is incorrect.

Anyway, I got a hold of arduino example code from https://www.banggood.com/1_54-Inch-or-2_13-Inch-or-2_9-Inch-Flexible-Ink-Screen-LCD-Module-E-paper-Electronic-Paper-Display-Compatible-with-Arduino-or-STM32-or-Raspberry-p-1811238.html?rmmds=myorder&cur_warehouse=CN&ID=63026666302660 and I intend to implement this in rust based on the traits from epd-waveshare.

So, finally my question. Is this something that should be added to this repo? or would it be better if I make my own lib that depends on epd-waveshare and publish that on my own github? (Or i guess a third possibility, that this is already supported somewhere i didnt find?)

caemor, 2023-08-30:
From my side, the display support can be added, the naming of the repository is no longer 100% correct regarding epd-*waveshare* anyway.
Just be aware that I am currently a bit behind in fixing the problems and pull requests.

caemor, 2024-10-28:
Closing this issue, but I am more than happy to accept PRs for non waveshare displays as well.

closed 2024-10-28 by caemor