rust-embedded-community/epd-waveshare #178: Cannot build example epd2in13_v2

theHausdorffMetric, 2023-11-19:
On an unchanged repo clone and using 
`stable-x86_64-unknown-linux-gnu unchanged - rustc 1.74.0 (79e9716c9 2023-11-13)`
Running `cargo  build --example epd2in13_v2` results in 
```
error[E0432]: unresolved import `linux_embedded_hal::Spidev`
  --> examples/epd2in13_v2.rs:19:22
   |
19 |     Delay, SPIError, Spidev, SysfsPin,
   |                      ^^^^^^
   |                      |
   |                      no `Spidev` in the root
   |                      help: a similar name exists in the module (notice the capitalization): `spidev`
   |
   = help: consider importing this struct instead:
           crate::spidev::Spidev

For more information about this error, try `rustc --explain E0432`.
error: could not compile `epd-waveshare` (example "epd2in13_v2") due to previous error
```

DaveRichmond, 2023-11-29:
So I just repaired this in pull request #179 but due to not having hardware to test this on right now I can't say for sure if anything else is broken

caemor, 2024-01-10:
Thanks!
This got fixed in #181 (and #179)

closed 2024-01-10 by caemor