rust-embedded-community/epd-waveshare #180: 2.9" Display - Partial Quick Refresh Support

scottdalgliesh, 2023-12-24:
I was able to get partial quick refresh working for the 2.9" display module by copying the source code for the 2.13" display and making only a couple of minor tweaks and changes (update buffer dimensions and set DisplayUpdateControl1 during initialization). I also incorporated the updated LUT array proposed in #150 to improve full refresh rate.

I'm not sure if there are technical reasons why the code for the 2.9" module was originally structured differently, but from my (perhaps naive) testing with both displays, this appears to make the 2.9" display work similarly to the 2.13" display, which is a significant improvement. If this is of interest, please let me know and I can open a PR with these modifications.

blinry, 2024-02-01:
As someone who owns a 2.9'' module, I'd be super interested! :) Is this for the v2?

scottdalgliesh, 2024-02-22:
> As someone who owns a 2.9'' module, I'd be super interested! :) Is this for the v2?

Yes, it is. I've been away for a bit, but will try to find time to upload the proof-of-concept changes to a public fork this weekend and add a link to it here. It looks like some recent changes have been made to this repo since my tests though, so I'll look at bringing it up to date before I upload it too.

scottdalgliesh, 2024-02-25:
Link to the proof-of-concept [Link](https://github.com/scottdalgliesh/epd-waveshare/tree/main)

I'm not sure if there is a technical reason why the 2.9" module code is currently structured differently (so use at your own risk!), but this has worked for me. Hope it helps!

caemor, 2024-10-28:
There is a faster LUT already available behind a feature gate for these "type a" driver displays. We grouped them together to reduce code duplication.

But I am happy to get a PR if it works better with your new changes.