rust-embedded-community/epd-waveshare #193: New release?

hackaugusto, 2024-02-25:
The last version on [crates.io](https://crates.io/crates/epd-waveshare/versions) is from 2 years ago, it would be great if there was a new release including the changes since then :)

KorribanMaster, 2024-03-11:
I would also love a new stable version to develop with as the embedded graphics package also had quite some changes as well

bemyak, 2024-03-12:
I've posted my changes here: https://github.com/caemor/epd-waveshare/pull/196
Hope the maintainers will have a chance to take a look before the next release :)

caemor, 2024-10-28:
Sorry for the delay. I released 0.6 with all the old changes. I didn't include #196 yet, because of the color change, which is something I need to think about first.

Another new release with all the other new features should hopefully arrive soon after.

closed 2024-10-28 by caemor