rust-embedded-community/epd-waveshare #195: How can I create a SpiDevice to initialize a WaveshareDisplay?

kongschlong, 2024-03-06:
_TL;DR: I need help with using this crate together with the `rp2040_hal`_

---

Hi everyone!

I am trying to connect a Waveshare 2.13" (v4) E-Paper display to a Raspberry Pi Pico using the `rp2040_hal`. Unfortunately I'm not able to build a working example since I'm having trouble understanding some of the types used by `epd-waveshare` and `embedded-hal`.

This is what my code looks like:

```rust

#[rp2040_hal::entry]
fn main() -> ! {
    let mut pac = pac::Peripherals::take().unwrap();
    let mut watchdog = hal::Watchdog::new(pac.WATCHDOG);

    // Configure the clocks
    let clocks = match hal::clocks::init_clocks_and_plls(
        XTAL_FREQ_HZ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    ) {
        Ok(r) => r,
        Err(_) => panic!("clocks init error")
    };

    let sio = hal::Sio::new(pac.SIO);

    // Set the pins to their default state
    let pins = hal::gpio::Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    let cs = pins.gpio9.into_push_pull_output();
    let busy = pins.gpio13.into_floating_input();
    let dc = pins.gpio8.into_push_pull_output();
    let rst = pins.gpio12.into_push_pull_output();
    // These are implicitly used by the spi driver if they are in the correct mode
    let spi_mosi = pins.gpio7.into_function::<hal::gpio::FunctionSpi>();
    let spi_miso = pins.gpio4.into_function::<hal::gpio::FunctionSpi>();
    let spi_sclk = pins.gpio6.into_function::<hal::gpio::FunctionSpi>();

    let spi = hal::spi::Spi::<_, _, _, 8>::new(pac.SPI0, (spi_mosi, spi_miso, spi_sclk));

    // Exchange the uninitialised SPI driver for an initialised one
    let mut spi = spi.init(
        &mut pac.RESETS,
        clocks.peripheral_clock.freq(),
        16.MHz(),
        embedded_hal::spi::MODE_0,
    );

    let mut timer = rp2040_hal::Timer::new(pac.TIMER, &mut pac.RESETS, &clocks);

    let mut display = Display2in13::default();
    let mut epd = Epd2in13::new(&mut spi, busy, dc, rst, &mut timer, None).expect("EPD setup error");

    loop {
        cortex_m::asm::wfi();
    }
}
```

The issue occurs when calling `Epd2in13::new()`: This method requires an `SPI` type that implements the `embedded_hal::spi::SpiDevice` trait.

The [example](https://github.com/caemor/epd-waveshare/blob/a083583441c166d07edaf2a6b5a31d025f45c483/examples/epd2in13_v2.rs#L30) uses a linux hal that opens a `SpidevDevice` for the SPI Interface. I assume that this `SpidevDevice` implements the required `SpiDevice` trait?

However, my `spi` variable has the type `rp2040_hal::spi::Spi`, which does not seem to implement `SpiDevice`, so I cannot pass it into the `Epd2in13::new()` function. I have to admit that I'm new to embedded Rust, and I'm not quite sure how the HALs work. So right now I'm stuck with the following question: How do I get a `SpiDevice` that I can use to communicate with the E-Paper display?

Maybe I'm missing something very simple that is obvious to most other users, but I think it could be helpful to add a few other examples to the crate that target some embedded microcontroller (STM32 or RP2040?), just to show how the connection to the display is established on different hardware platforms.

Thanks in advance!

---

For reference, these are the crates I am using:

```toml
[dependencies]
#epd-waveshare = "0.5.0"  # the version on crates.io is outdated
epd-waveshare = { git = "https://github.com/caemor/epd-waveshare.git" }
embedded-graphics = "0.8.1"
chrono = { version = "0.4.33", default-features = false }

defmt = "0.3.6"
defmt-rtt = "0.4.0"

rp2040-hal = { version = "0.9.2", features = ["eh1_0_alpha", "defmt", "critical-section-impl"]}
rp2040-boot2 = "0.3.0"
embedded-hal = "1.0.0"
cortex-m = "0.7.7"
cortex-m-rt = "0.7.3"
panic-halt = "0.2.0"
```

andber1, 2024-03-11:
It seems that only a SpiBus is implemented in rp-hal, not a SpiDevice. See [here](https://github.com/rp-rs/rp-hal/pull/488). You can implement a SpiDevice yourself. The [embedded-hal-bus](https://docs.rs/embedded-hal-bus/0.1.0/embedded_hal_bus/) crate provides mechanisms to connect a SpiBus and a SpiDevice. The [ExclusiveDevice](https://docs.rs/embedded-hal-bus/0.1.0/embedded_hal_bus/spi/struct.ExclusiveDevice.html) seems to be the easiest way.

kongschlong, 2024-03-12:
Hi! Thank you so much, that really helped me out.

Unfortunately I'm still not fully able to compile my project because I'm not sure how the `DC` pin is used. My assumption is that it is the `MISO` part of the SPI connection, so it makes sense that `Spi::new()` takes ownership of this pin. However, `Epd2in13::new()` also expects access to this DC pin, which is not possible because it was moved into `Spi` before.

Here's a snippet of the relevant part of my program:

```rust

#[rp2040_hal::entry]
fn main() -> ! {
   // [...]

    // Set the pins to their default state
    let pins = hal::gpio::Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    let pin_cs = pins.gpio14.into_push_pull_output();
    let pin_busy = pins.gpio12.into_floating_input();
    let pin_dc = pins.gpio8.into_push_pull_output();
    let pin_rst = pins.gpio13.into_push_pull_output();
    // These are implicitly used by the spi driver if they are in the correct mode
    let spi_mosi = pins.gpio11.into_function::<hal::gpio::FunctionSpi>(); // DIN
    // This doesn't work, and it shouldn't. (It doesn't make sense to initialize the same pin twice)
    // But how do I pass the same pin into `Spi` and into `Epd2in13`?
    let spi_miso = pins.gpio8.into_function::<hal::gpio::FunctionSpi>(); // DC
    let spi_sclk = pins.gpio10.into_function::<hal::gpio::FunctionSpi>();

    let spi = hal::spi::Spi::<_, _, _, 8>::new(pac.SPI1, (spi_mosi, spi_miso, spi_sclk));

    // Exchange the uninitialised SPI driver for an initialised one
    let mut spi = spi.init(
        &mut pac.RESETS,
        clocks.peripheral_clock.freq(),
        16.MHz(),
        embedded_hal::spi::MODE_0,
    );

    let mut timer = rp2040_hal::Timer::new(pac.TIMER, &mut pac.RESETS, &clocks);

    let mut spi_device = ExclusiveDevice::new(spi, pin_cs, timer);

    info!("Setting up display");

    let mut display = Display2in13::default();
    let mut epd = Epd2in13::new(&mut spi_device, pin_busy, pin_dc, pin_rst, &mut timer, None).expect("EPD setup error");

    epd.update_frame(&mut spi_device, display.buffer(), &mut timer).expect("Update frame");
    epd.display_frame(&mut spi_device, &mut timer).expect("Display new frame");

    loop {
        cortex_m::asm::wfi();
    }
}
```

So basically my question is: How can I pass the same DC pin into both the Spi driver and the EPD driver?

andber1, 2024-03-13:
The pins MOSI, MISO and SCLK make the SPI bus. Together with CS (chip select) this is the SPI device. EPD needs the SPI device, BUSY, DC (to distinguish between data and control bytes, has nothing to do with SPI) and RESET. These are all different pins. Maybe the confusion is that most e-paper modules do not have a MISO pin (master in, slave out), because the e-paper does not need to send data to the host. You can leave this pin unconnected. MOSI (master out, slave in) is often described as DIN (data in) or SDI (slave data in).

kongschlong, 2024-03-14:
Thank you so much for your help, it's finally working now (well, the colors are inverted and the screen is rotated 90 degrees, but I'm sure I'll figure that out later).

---

For future reference, the following pin mapping is tested on a Raspberry Pi Pico connected to the Waveshare 2.13" Display (Rev. 4) using the universal Driver HAT.

| RPi Pico | EPD | Notes |
|---|---|---|
| 8 | DC | |
| 9 | BUSY | |
| 10 | CLK | SCK pin of SPI bus |
| 11 | DIN | MOSI pin of SPI bus |
| 12 | not connected | MISO pin of SPI bus |
| 13 | RST | |
| 14 | CS | Chip Select for SPI bus

`Display Config` switch: B
`Interface Config` switch: 4-line SPI

closed 2024-03-14 by kongschlong