rust-embedded-community/epd-waveshare #198: Waveshare 1.54 V2 does not work

agmes4, 2024-05-06:
# Problem
I tried using this lib for the waveshare 1.54 V2 this did not work 

## Things I noticed 
After some trying I noticed that when ever writing data image in 4 Pin mode the chip select is pulled to high every time a byte was written to the bus. When trying an other Arduino sketch it worked. The only change I noticed on the Bus was that it did not pull the CS to high when ever a byte of the image was written.

semiversus, 2024-08-09:
I have the same problem and can confirm the strange chip select behavior.

caemor, 2024-10-28:
Could you try to change the `SINGLE_BYTE_WRITE` to `false` in `src/epd1in54_v2/mod.rs:12`.
I don't have the same display in my hands so I'm not able to test this.

ImplFerris, 2025-03-10:
> Could you try to change the SINGLE_BYTE_WRITE to false in src/epd1in54_v2/mod.rs:12.
I just tried it now, it didn't work for me.

ImplFerris, 2025-03-11:
I tried to pinpoint the exact issue and map it to the datasheet but couldn't find the exact problem. I ended up with a hacky solution.

https://github.com/ImplFerris/epd-waveshare/tree/1in54_v2_hacky_fix

ImplFerris, 2025-04-17:
i kind of have a hacky fix
https://github.com/ImplFerris/epd-waveshare