rust-embedded-community/epd-waveshare #223: e-paper 7.5" doesn't work?

DaAitch, 2024-12-05:
I have 3 displays for my esp32 e-paper driver board:

- epd1in54_v2 => worked
- epd2in13_v2 => worked
- e-paper 7.5" (B) v3 => doesn't work, i tried all epd7in5_xxx versions

The 7.5inch doesn't work and the program crashes. Does somebody have an idea?

Thank you in advance.

```log
I (473) main_task: Calling app_main()
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
0x40080400 - _invalid_pc_placeholder
    at ??:??
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
I (61) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (61) boot: compile time Jun  7 2023 07:48:23
I (64) boot: Multicore bootloader
I (68) boot: chip revision: v1.0
I (72) boot.esp32: SPI Speed      : 40MHz
I (77) boot.esp32: SPI Mode       : DIO
I (81) boot.esp32: SPI Flash Size : 4MB
W (86) boot.esp32: PRO CPU has been reset by WDT.
W (91) boot.esp32: WDT reset info: PRO CPU PC=0x40082f25
0x40082f25 - panicHandler
    at /Users/aitch/Code/esp32_rs/.embuild/espressif/esp-idf/v5.2.2/components/esp_system/port/panic_handler.c:217
W (97) boot.esp32: WDT reset info: APP CPU PC=0x40111236
0x40111236 - panic_handler
    at /Users/aitch/Code/esp32_rs/.embuild/espressif/esp-idf/v5.2.2/components/esp_system/port/panic_handler.c:138
I (103) boot: Enabling RNG early entropy source...
I (109) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
I (120) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (127) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (135) boot:  2 factory          factory app      00 00 00010000 003f0000
I (143) boot: End of partition table
I (147) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2435ch (148316) map
I (209) esp_image: segment 1: paddr=00034384 vaddr=3ffb0000 size=0251ch (  9500) load
I (213) esp_image: segment 2: paddr=000368a8 vaddr=40080000 size=09770h ( 38768) load
I (231) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=55740h (350016) map
I (358) esp_image: segment 4: paddr=00095768 vaddr=40089770 size=04290h ( 17040) load
I (372) boot: Loaded app from partition at offset 0x10000
I (372) boot: Disabling RNG early entropy source...
I (383) cpu_start: Multicore app
I (392) cpu_start: Pro cpu start user code
I (392) cpu_start: cpu freq: 160000000 Hz
I (392) cpu_start: Application information:
I (395) cpu_start: Project name:     libespidf
I (400) cpu_start: App version:      cf567d5-dirty
I (406) cpu_start: Compile time:     Dec  5 2024 18:26:23
I (412) cpu_start: ELF file SHA256:  000000000...
I (417) cpu_start: ESP-IDF:          v5.2.2
I (422) cpu_start: Min chip rev:     v0.0
I (427) cpu_start: Max chip rev:     v3.99 
I (432) cpu_start: Chip rev:         v1.0
I (437) heap_init: Initializing. RAM available for dynamic allocation:
I (444) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (450) heap_init: At 3FFB3268 len 0002CD98 (179 KiB): DRAM
I (456) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (462) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (469) heap_init: At 4008DA00 len 00012600 (73 KiB): IRAM
I (476) spi_flash: detected chip: generic
I (479) spi_flash: flash io: dio
W (484) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (492) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (503) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (512) main_task: Started on CPU0
I (522) main_task: Calling app_main()
```

here is my Rust code for ESP32 driver board with the displays:

```rs

type Display = epd_waveshare::epd7in5b_v3::Display7in5;
type Epd<SPI, BUSY, DC, RST, DELAY> =
    epd_waveshare::epd7in5b_v3::Epd7in5<SPI, BUSY, DC, RST, DELAY>;
type Col = TriColor;

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let p = Peripherals::take().unwrap();

    let mut device = SpiDeviceDriver::new_single(
        p.spi2,
        p.pins.gpio13,
        p.pins.gpio14,
        Option::<AnyIOPin>::None,
        Some(p.pins.gpio15),
        &SpiDriverConfig::default(),
        &Config::default(),
    )
    .unwrap();

    let mut delay = Ets;

    let busy = PinDriver::input(p.pins.gpio25).unwrap();
    let dc = PinDriver::output(p.pins.gpio27).unwrap();
    let rst = PinDriver::output(p.pins.gpio26).unwrap();

    let mut display = Display::default();
    let mut epd =
        Epd::new(&mut device, busy, dc, rst, &mut delay, None).expect("e-ink initalize error");

    display.clear(Col::White).unwrap();

    draw_text(&mut display, "works!", 0, 0);
    epd.update_frame(&mut device, display.buffer(), &mut delay)
        .unwrap();
    epd.display_frame(&mut device, &mut delay).unwrap();

    epd.sleep(&mut device, &mut delay).unwrap();
}

fn draw_text(display: &mut Display, text: &str, x: i32, y: i32) {
    let style = MonoTextStyleBuilder::new()
        .font(&embedded_graphics::mono_font::ascii::FONT_10X20)
        .text_color(Col::Black)
        .background_color(Col::White)
        .build();

    let text_style = TextStyleBuilder::new().baseline(Baseline::Top).build();

    let _ = Text::with_text_style(text, Point::new(x, y), style, text_style).draw(display);
}
```

bemyak, 2024-12-05:
It seems you're running out of stack space and need to allocate the display buffer on heap. See this comment: https://github.com/caemor/epd-waveshare/blob/6992771d0c3d01b8d402ed2253c9f6d290556faa/src/traits.rs#L130

DaAitch, 2024-12-06:
Hey, wow, thank you. At least the screen shows the result, but

1. it's flickering a lot for 10s until it stabilizes with the correct result
2. it seems like the tasks are too heavy so that the watch dog times out (I guess due to the screen size)
3. the first error `_invalid_pc_placeholder` looks like there is another issue before. Even boxing will create the display on the stack and move it then. Is there maybe another problem where I need to use unsafe Rust `Box::<Display>::new_uninit()` to not use stack at all for it.

Here is the [output as gist](https://gist.github.com/DaAitch/11fbf816fa95551b7a55ca463f8449ad).

@bemyak can you tell me how you find out that I ran out of stack? I'm unable to read the error message, but would like to understand it.

DaAitch, 2024-12-06:
Despite the last `_invalid_pc_placeholder` the rest is clear to me now:

- adding `CONFIG_ESP_TASK_WDT_EN=n` to `sdkconfig.defaults` will disable watch dog
- according to the `Selection Guide` on waveshare my display takes 26s and my stopwatch confirmed that, but it looks very strange how it flickers with different frequencies.